raspberrypi/bookworm-feedback #421: wf-panel-pi leaks about 19 MiB of memory / 24 hours

Consolatis, 2025-06-11:
I noticed the `wf-panel-pi` process using about 1 GiB of RSS after about 50 days uptime. Killing the process fixed it and made it consume about 62 MiB. Then I started to log virtual memory, residential memory and shared memory at some random points in time for a few days and did some calculations.

The conclusion is:
RSS increases linearly with a rate of about `0.79` MiB every hour or about 19 MiB every day. The virtual memory mappings increase along with it which indicates this is something heap allocated. The shared memory usage is rock stable at the initial value.

I didn't interact with the panel at all, the system just sat idle (with a monitor connected but shut down, wlr-randr shows it as connected) and occasional wayvnc connections.

spl237, 2025-06-12:
OK, thanks - will look into what is happening. Was this on the latest released RPiOS image?

Consolatis, 2025-06-12:
> Was this on the latest released RPiOS image?

No, it is an older image but I did `apt update && apt upgrade` a few times.

> Edit - with 50 days uptime, this was presumably an older version of wf-panel. Had you updated the system since it was booted?

Yes, the tests were done using the up-to-date version from the repo (`0.101`).
The only packages not up-to-date during the tests were:
```
$ apt list --upgradable
gstreamer1.0-plugins-bad/stable-security 1.22.0-4+deb12u6 arm64 [upgradable from: 1.22.0-4+deb12u5]
libgstreamer-plugins-bad1.0-0/stable-security 1.22.0-4+deb12u6 arm64 [upgradable from: 1.22.0-4+deb12u5]
python3-pidng/stable 4.0.9+git20250603.8427835-1 arm64 [upgradable from: 4.0.9+git20250121.e2f5a8d-1]
rpi-eeprom/stable,stable 28.2-1 all [upgradable from: 28.1-1]
```

Consolatis, 2025-06-13:
> And did you have just the standard set of plugins installed, or had you added any of the CPU or GPU monitors?
>> Missed the question about the standard set of plugins. The panel config is completely unmodified, wlan is active.

I am sorry, I must have added the resource monitors to the panel directly after installing the image and then forgot about me adding them. I assumed they were part of the default config but deleting `wf-panel-pi.ini` and restarting the panel doesn't show them.

I ran without the clock plugin for some hours and it didn't make a difference. Then I removed the monitors and now RES is rock stable so it is something related to one of the resource monitors. I'll test them one by one to see if its just a specific one that is affected or all of them.

spl237, 2025-06-15:
Just done a quick test, and with all 3 resource monitors, I do see a leak. Now trying them one at a time to see if it is one specific plugin, or the shared graph drawing code.

spl237, 2025-06-15:
OK, I've found a potential leak in the graph drawing code, which would affect all three plugins. Have fixed that and am re-testing to see if things have improved.

Consolatis, 2025-06-15:
I had one resource monitor running for a while and it was approximately a third of the previous leak so this checks out.

> I've pushed the fix into our original bookworm branch as well as the trixie code - will release a fixed version tomorrow. If you then get a chance to confirm it has fixed the issue for you, that would be great.

Sure thing, thanks!

Consolatis, 2025-06-16:
Yes it is. Updated, added all 3 resource monitors, killed the process and noted down the start RES value. I'll keep an eye on it for the next hours.

Consolatis, 2025-06-16:
Results after about 9 hours: within the first hour, RES has risen of a total of about 1 MiB (which I assume is ~333 KiB per resource monitor to fill the initial data point history). After that it is rock stable.

So I consider this fixed. Thanks again!

closed 2025-06-16 by Consolatis

spl237, 2025-06-16:
Excellent - many thanks for testing, and for the original report!