raspberrypi/bookworm-feedback #422: Interlaced output on Lite versions of Raspbian OS broken

andytij, 2025-06-14:
I have some Pis that are video players showing feeds in a broadcaster, where the video environment is a strict 1920x1080 50 i (**Interlaced**).

I am attempting to update from Buster to the latest, but have noticed that interlaced output stopped working, possibly between February and May 2023.

I am using the Lite version or the raspberry pi OS - console view, the setting should be set in `/boot/firmware/cmdline.txt` in Bookworm, and `/boot/cmdline.txt` in Bullseye with a line something along the lines of `video=HDMI-A-1:1920x1080@50i`

However, I keep getting **progressive** pictures out of the Pi with the bookworm images!

I tried seeing at which point it broke by downloading Images from the Raspberry Pi OS website. It looks like the setting last worked in [https://downloads.raspberrypi.com/raspios_lite_arm64/images/raspios_lite_arm64-2023-02-22/](url) - but if you apt upgrade it breaks.

It stopped working in the OS image from the next version which was the last Bullseye version [https://downloads.raspberrypi.com/raspios_lite_arm64/images/raspios_lite_arm64-2023-05-03/](url)

I'm not sure if it is helpful, but one of these packages breaks it:
`adduser avahi-daemon base-files bind9-host bind9-libs bluez bluez-firmware bsdextrautils bsdutils busybox cifs-utils cpio curl dbus debian-archive-keyring distro-info-data dns-root-data dnsmasq-base dpkg dpkg-dev e2fsprogs eject exfatprogs fdisk file firmware-atheros firmware-brcm80211 firmware-libertas firmware-misc-nonfree firmware-realtek grep isc-dhcp-client isc-dhcp-common less libavahi-common-data libavahi-common3 libavahi-core7 libblas3 libblkid1 libbluetooth3 libbpf0 libbsd0 libc-bin libc-dev-bin libc-devtools libc-l10n libc6 libc6-dbg libc6-dev libcamera-apps-lite libcamera0 libcap2 libcap2-bin libcom-err2 libcurl3-gnutls libcurl4 libdbus-1-3 libdpkg-perl libebml5 libexpat1 libext2fs2 libfdisk1 libflac8 libfreetype-dev libfreetype6 libfreetype6-dev libglib2.0-0 libgnutls30 libgssapi-krb5-2 libgstreamer-plugins-base1.0-0 libgstreamer1.0-0 libjson-c5 libk5crypto3 libkrb5-3 libkrb5support0 liblapack3 libmagic-mgc libmagic1 libmount1 libncurses6 libncursesw6 libndp0 libnftables1 libnghttp2-14 libnss3 libntfs-3g883 libpam-systemd libperl5.32 libpython3.9 libpython3.9-minimal libpython3.9-stdlib libraspberrypi-bin libraspberrypi-dev libraspberrypi-doc libraspberrypi0 libsepol1 libsmartcols1 libsqlite3-0 libss2 libssh2-1 libssl1.1 libsystemd0 libtasn1-6 libtiff5 libtinfo6 libudev1 libuuid1 libuv1 libwbclient0 libwebp6 libwebpdemux2 libwebpmux3 libx11-6 libx11-data libxml2 libxpm4 linux-libc-dev locales login logrotate logsave mount nano ncurses-base ncurses-bin ncurses-term net-tools nfs-common nftables ntfs-3g openssh-client openssh-server openssh-sftp-server openssl passwd perl perl-base perl-modules-5.32 python3-idna python3-libcamera python3-picamera2 python3-pil python3-pkg-resources python3-urllib3 python3-v4l2 python3.9 python3.9-minimal raspberrypi-bootloader raspberrypi-kernel raspberrypi-sys-mods raspi-config rfkill rpi-eeprom rsync ssh systemd systemd-sysv systemd-timesyncd tar tzdata ucf udev util-linux vim-common vim-tiny wget wireless-regdb wpasupplicant xxd`

6by9, 2025-06-16:
Tested on a Pi4 running Raspberry Pi OS Bookworm with Dell 4k monitor connected to HDMI-A-1 with `video=HDMI-A-1:1920x1080@50i` in /boot/firmware/cmdline.txt. The monitor reports 1080i as expected.

kmsprint returns
```
pi@usb-bookworm:~ $ kmsprint
Connector 0 (33) HDMI-A-1 (connected)
  Encoder 0 (32) TMDS
    Crtc 3 (100) 1920x1080i@50.00 74.250 1920/528/44/148/+ 1080/4/10/31/+ 50 (50.00) U|D 16:9
      Plane 3 (89) fb-id: 719 (crtcs: 3) 0,0 1920x1080 -> 0,0 1920x1080 (XR24 AR24 AB24 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU24 YV24 YU12 YV12 NV12 NV21 NV16 NV61 P030 XR30 AR30 AB30 XB30 RGB8 BGR8 XR12 AR12 XB12 AB12 BX12 BA12 RX12 RA12)
        FB 719 1920x1080 RG16
Connector 1 (42) HDMI-A-2 (connected)
  Encoder 1 (41) TMDS
    Crtc 4 (112) 1920x1080@60.00 148.500 1920/88/44/148/+ 1080/4/5/36/+ 60 (60.00) P|D 
      Plane 4 (101) fb-id: 719 (crtcs: 4) 0,0 1920x1080 -> 0,0 1920x1080 (XR24 AR24 AB24 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU24 YV24 YU12 YV12 NV12 NV21 NV16 NV61 P030 XR30 AR30 AB30 XB30 RGB8 BGR8 XR12 AR12 XB12 AB12 BX12 BA12 RX12 RA12)
        FB 719 1920x1080 RG16
```
(There's a 1080p monitor on HDMI-2).

1080i50 is in the EDID for this 4k monitor. It should get added using CVT timings if it isn't, but those may not be the timings you're after (CEA or DMT are generally more useful).

There is something odd as using `video=HDMI-A-2:1920x1080@50i` is giving me 1080p50 on that second monitor (Hannspree HT225). The kernel is putting the userdefined flag against the wrong mode (1080p50). I'm just looking into that one.

Which version of Pi are you using, and please provide the EDID of the connected display (easiest with `edid-decode /sys/class/drm/card1-HDMI-A-1/edid`)

andytij, 2025-06-18:
Thanks @6by9 
Did you still need me to run the EDID command?

andytij, 2025-06-19:
I'm glad it was an easy one @6by9 

Thanks again
Andy

lurch, 2025-06-19:
@andytij Can this be closed? (on the assumption that @6by9 's kernel fix will eventually trickle through into an APT-installed kernel)

closed 2025-06-19 by andytij