raspberrypi/bookworm-feedback #427: Bookworm Lite seems to have issues on 256MB boards

andrum993, 2025-06-27:
Just tried on install on a rev 1 Raspberry Pi 1 board with 256MB of RAM. When I tried updating the packages I got an error related to the /run filesystem being nearly full, which caused the upgrade process to fail. I'm filing this issue really to see if the 256MB boards are actually still supported by Raspberry Pi, or whether they are now EOL in terms of support in the current version of the OS.

See https://forums.raspberrypi.com/viewtopic.php?p=2322907#p2322907 for a bit more information about the error.

XECDesign, 2025-06-27:
They're not EOL and we should support them. We can't support a full desktop and browser experience anymore, but you should definitely be able to update/upgrade.

Thanks for reporting the issue.

andrum993, 2025-06-27:
This was just something I came across in passing, as I was curious if these boards where still supported. It's not something I need to actually work, as I took my sole 256MB board out of service a few years ago.

I think apt update/upgrade used to work correctly on Bookworm on the 256MB boards when Raspberry Pi OS Bookworm was first released, as I think I tried this early on in Bookworm's life. I suspect this got broken somewhere along the way by something Debian have done. Or I may be misremembering this, and it was actually Bullseye that worked.

andrum993, 2025-06-28:
Did some more testing. I increased the tmpfs allocation from 16MB, which is what it gets set to by default on my 256MB board, to 32MB using an entry in /etc/fstab. First boot after making that change causes a hang late in boot. Second boot it pauses a few times late in boot but gets to the login prompt and then seems to work fine. Third boot and it's back to working as it was before making the change. I then attempted the sudo apt update, sudo apt full-upgrade which failed with the out-of-the-box setup. Obviously this is rather slow, and I can see the swap usage going up as it runs, but it now works and completes as expected.

Given the difficulties I've found with my 256MB board, plus the fact that there are likely to be only a very small number of these boards out there, combined with the fact that Trixie is on the horizon, and there doesn't seem to be anyone piping up on the forums to complain about their 256MB board not working, you might be best just EOLing these boards, or if anyone asks, just say someething like "the 256MB boards don't quite have enough memory to run current releases of Raspberry Pi OS properly. It may be possible for you to get it working with some customisation, but it's not something we will be able to help you with".

I'm closing this issue as my query has been answered by XECDesign above, and I've satisfied my curiosity into whether these boards actually work or not.

Consolatis, 2025-06-28:
256 MiB ram should be plenty for a simple CLI setup with some services like nginx or whatever. There is also the option to use zram backed swap.

roliver-rpi, 2025-07-03:
I believe the upgrade of libc6 is failing as systemd is unable to reexec itself. It's possible to reproduce this issue without invoking apt at all:
```
$ sudo systemctl daemon-reexec
Reexecute daemon failed: Refusing to reexecute, not enough space available on /run/systemd. Currently 15.6M are free, but a safety buffer of 16.0M is enforced.
```

Although it looks like the [16.0M limit is arbitrary](https://github.com/systemd/systemd/issues/5016#issuecomment-870772760), it's not possible to lower the hard-coded `RELOAD_DISK_SPACE_MIN` limit at runtime (anecdotal evidence suggests that 1.0M would be sufficient for embedded systems).

Given that the systemd source explicitly calls for a 20% RAM '/run' on 256MB devices (see TMPFS_LIMITS_RUN in systemd-252.38 source), we should consider changing initramfs-mounted /run to 20% (rather than its 10% default). This is currently set as RUNSIZE=10% in /etc/initramfs-tools/initramfs.conf (initramfs-tools-core package). Alternatively, initramfs.runsize=20% could be given on the cmdline. I have been able to 'full-upgrade' the 2025-05-13 Lite image with this setup.

gamefreak613, 2025-07-20:
Just wanted to add that I ran into this issue. Been using my Pi 1 for ages as a VPN server. Unfortunately it has stopped working and so I went to go re-implement, and I'm running into this issue.

XECDesign, 2025-07-21:
Just shipped a new version of `initramfs-tools` with RUNSIZE=20% by default. It doesn't help users running the current image, but should be resolved with the next one.

Until then, you can manually edit `/etc/initramfs-tools/initramfs.conf` and then run `sudo update-initramfs -u -k all` and reboot.

closed 2025-07-21 by XECDesign