raspberrypi/bookworm-feedback #428: Removal of `libraspberrypi` packages on system upgrade

cmitu, 2025-07-02:
Hi,

the recent version of the `raspi-utils-dt` package has added a _replaces_ for `libraspberrypi-bin`, which makes it's upgrade to remove `libraspberrypi0`/`libraspberrypi-bin`/`libraspberrypi-dev`. Is this intended ? 

The `libraspberrypi` package contain some files which I don't think they're offered by other packages (though not so sure it's useful in recent RaspiOS versions).

lurch, 2025-07-02:
@cmitu IIRC this isn't a problem if you start from a recent Bookworm image; it's only a problem if you upgrade a really old Raspberry Pi OS Bookworm image? I also ran into something similar recently, and searching through my bash history, I _think_ the solution might have been:
```
sudo dpkg -r libraspberrypi0 libraspberrypi-dev libraspberrypi-doc
```
(note that you **have** to use `dpkg` here, you can't use a higher-level tool like `apt-get` or `apt`).

ping @XECDesign for verification and/or additional information 🙂

cmitu, 2025-07-02:
It's true I have an 'old' image installed, periodically upgraded, but I didn't have an issue with `apt full-upgrade` to upgrade the package - it removed the `libraspberrypi` packages when upgrading `raspi-utils-dt`.

_However_, I did have some 3rd party package(s) that depedended on `libraspberrypi0` which caused in the first phase for the `raspi-utils-dt` package to not be upgraded automatically. Looking at the dependency issue I noticed that the 3rd party package was blocking the (full-) upgrade and - sure enough - once I removed them, the upgrade of `raspi-utils-dt` went ok (removing the `libraspberrypi` packages installed sometime ago as depedency to the other package).

lurch, 2025-07-02:
Ahhh, can this issue be closed then?

cmitu, 2025-07-02:
No, I'd like to know whether this is intentional or not - in order to change the dependency for the 3rd party packages.

lurch, 2025-07-02:
Would you mind sharing details about what that 3rd-party package is? Maybe we'll be able to suggest what alternative package they ought to be depending on instead.

lurch, 2025-07-02:
Correct; as you're aware the old display interfaces are no longer supported. Instead, Raspberry Pi now supports more standardised Linux display APIs (DRM, OpenGL, GLES, Vulkan, Wayland, etc.), which means those emulators need less Raspberry Pi-specific code 🕹 🎮  👾  😃