raspberrypi/bookworm-feedback #429: HDMI multichannel audio. Raspberry Pi OS for Raspberry Pi 5 came so close to becoming a full-featured multimedia center BUT...

eitijupaenoithoowohd, 2025-07-03:
Raspberry Pi OS could have been the perfect OS for a multimedia center. It has everything needed to fully play 4K 60Hz video with hardware acceleration, audio through PulseAudio or PipeWire based on user preference, but unfortunately the audio is limited to 2.0 and lacks multichannel audio support through HDMI. This is very strange because LibreELEC runs perfectly on the same Raspberry Pi 5 hardware platform and supports multichannel audio through raw stream passthrough via HDMI.

How they do it: [2.0 audio with ALSA, activate passthrough, enable AC3 and DTS.](https://forum.libreelec.tv/thread/28021-raspberry-pi-5-audio-passthrough/)

Configuration Examples:

* Real-time A52/AC3 encoding configuration
~/.asoundrc file for using a Raspberry Pi with onboard audio for output and a webcam (or other device) as audio input. Both combined in a virtual ALSA device. · GitHub https://forums.raspberrypi.com/viewtopic.php?t=172536

* HDMI passthrough setup
Passthrough multichannel audio over HDMI - Raspberry Pi Forums https://forums.raspberrypi.com/viewtopic.php?t=221734

* GitHub examples
   * Example .asoundrc for audio output recording
    * Asoundrc - AlsaProject  https://gist.github.com/julthon/12f056dff0135cfb56baf911067aa11b

Unfortunately, my knowledge and understanding of hardware architecture isn't sufficient to repeat these settings for Raspberry Pi OS. Perhaps someone can figure this out and help with it?

eitijupaenoithoowohd, 2025-07-03:
> To start with, run raspinfo | pastebinit -b dpaste.com to report your current config.

This is essentially the default configuration: https://dpaste.com/6VVSR6KEE

6by9, 2025-07-03:
The output of `raspinfo` includes the EDID from your monitor.

And that reveals that your Samsung 3578 only supports stereo.
```
Audio Data Block:
    Linear PCM:
      Max channels: 2
      Supported sample rates (kHz): 48 44.1 32
      Supported sample sizes (bits): 24 20 16
  Speaker Allocation Data Block:
    FL/FR - Front Left/Right
```
There's little point in trying to send multi-channel audio over HDMI if the receiving device doesn't support it.

eitijupaenoithoowohd, 2025-07-04:
I turned on the soundbar. Installed Kodi on RPiOS. Tried running multichannel tests from within Kodi. Went through all available audio settings. Useless. Complete silence from the speakers.

Here's the new raspinfo captured at that moment. I didn't notice any significant difference:
https://dpaste.com/9ZN2CHW78

6by9, 2025-07-04:
A splitter totally changes the problem.

Generally a splitter is used to drive multiple monitors with the same source, and therefore they rewrite the EDID with the _subset_ of features that are supported by all connected devices. That subset would be basic stereo audio when your TV and soundbar are connected. In my experience they become more fun as the EDID rewriting isn't always totally deterministic and connecting devices in different orders results in different EDIDs.

An AV Receiver (AVR) or soundbar would normally be connected to the source device first so that it has the chance to rewrite the EDID and _add_ the relevant audio modes that it can strip off before passing the video to the display.
ARC and eARC are further complications as the TV is then advertising the audio capabilities via the EDID, and then forwarding the audio signal to the AVR or soundbar. I don't recall the exact mechanism that it uses to determine what audio modes can be supported by the audio end of the ARC channel.

If Libreelec is doing something different then we'd need to see logs from that - see https://wiki.libreelec.tv/support/log-files#libreelec-settings for generating logs.

eitijupaenoithoowohd, 2025-07-06:
To completely eliminate any potential errors that intermediary devices might introduce, I disabled the splitter and connected the HDMI output from the Raspberry Pi 5 directly to the HDMI input of the soundbar. The soundbar has a passthrough output to the TV. Yes, sir, that option is available! Then I formatted the MicroSD card and reinstalled Raspberry Pi OS from scratch.

Here's the fresh device log dump: https://dpaste.com/GVEVTK86V

However, none of this helped resolve the multichannel audio issue. This is the default audio output device I see:

[image]

These are the audio output devices I see when I switch the VLC settings to ALSA:

```
Discard all samples (playback) or generate zero samples (capture)
Playback/recording through the PulseAudio sound server
Default Audio Device
Rate Converter Plugin Using Libav/FFmpeg Library
Rate Converter Plugin Using Samplerate Library
Rate Converter Plugin Using Speex Resampler
JACK Audio Connection Kit
Open Sound System
PulseAudio Sound Server
Plugin using Speex DSP (resample, agc, denoise, echo, dereverb)
Plugin for channel upmix (4,6,8)
Plugin for channel downmix (stereo) with a simple spacialization
vc4-hdmi-0, MAI PCM i2s-hifi-0 Direct hardware device without any conversions
vc4-hdmi-0, MAI PCM i2s-hifi-0 Hardware device with all software conversions
vc4-hdmi-0, MAI PCM i2s-hifi-0 Default Audio Device
vc4-hdmi-0, MAI PCM i2s-hifi-0 HDMI Audio Output
vc4-hdmi-0, MAI PCM i2s-hifi-0 Direct sample mixing device
vc4-hdmi-0 USB Stream Output
vc4-hdmi-1, MAI PCM i2s-hifi-0 Direct hardware device without any conversions
vc4-hdmi-1, MAI PCM i2s-hifi-0 Hardware device with all software conversions
vc4-hdmi-1, MAI PCM i2s-hifi-0 Default Audio Device
vc4-hdmi-1, MAI PCM i2s-hifi-0 HDMI Audio Output
vc4-hdmi-1, MAI PCM i2s-hifi-0 Direct sample mixing device
```

I want to draw your attention to the fact that I tested literally every single device from this list, and I only hear 2.0 audio. For multichannel audio testing, I downloaded these Fraunhofer samples:

https://www2.iis.fraunhofer.de/AAC/multichannel.html

I also installed Kodi from the packages, but it doesn't even have the option to switch to ALSA.

After that, I booted from another MicroSD card with LibreELEC and ran the same Fraunhofer samples to test audio with direct connection WITHOUT the splitter. This time, multichannel audio worked just as well under LibreELEC as it did with the splitter. Please review this LibreELEC log dump:

https://dpaste.com/DFP3AAVG5