raspberrypi/bookworm-feedback #431: Latest updates made VNC unusable

IceCastleMatt, 2025-07-08:
I have a Pi 4 Model B+ running Bookworm that we are using to automate/remotely control some critical functions on our jobsite.  We mainly access it via VNC or Raspberry Pi Connect, but we have a monitor connected (in portrait mode, since we mainly remote-in on our phones) so that the desktop will display consistently across all of our devices.  We have also given VNC users discreet usernames and passwords, so that they can access shortcuts on the main desktop without needing to give out the root login.

Yesterday, after installing the latest updates, I was only able to log in using the root account's credentials; all other usernames and passwords were rejected as invalid.  This is obviously a problem, as we need multiple users to have access to the desktop without having root access.

When I did remote-in with the root account, the cursor on the desktop was sideways (as if the desktop were in portrait mode but the cursor was oriented landscape) and there was a substantial offset between where the cursor was displayed and what it was actually highlighting (for instance, in the "Start" menu, I would have to hover the cursor two menu items below the one I wanted to select).  This offset went away when I changed the screen's orientation to landscape, but this screen really needs to be in portrait mode for our purposes.

any1, 2025-07-08:
If you edit `/etc/pam.d/wayvnc` and remove the middle line (`pam_allow_desktopuser.so`), you should be able to log in as any user again.

The rotated cursor is a bug in wayvnc. It receives the buffer rotated from labwc and needs to undo the rotation before sending it. I completely forgot to test that when I implemented the client-side cursor stuff.

any1, 2025-07-23:
Here is a fix for the incorrectly rotated cursor: https://gitlab.freedesktop.org/wlroots/wlroots/-/merge_requests/5120

IceCastleMatt, 2025-09-06:
So...the cursor issue came back with the latest wayvnc update  (both rotation and offset).

Last time, any1's fix (adding the `--render-cursor` argument to `wayvnc` in `/usr/sbin/wayvnc-run.sh`) did the trick.  I tried re-adding it after this update, but this time it didn't work.

Interestingly, I don't experience the issue on Raspberry Pi Connect; only on local VNC clients (TigerVNC and RealVNC).

closed 2025-09-07 by IceCastleMatt