raspberrypi/bookworm-feedback #442: Can't change keyboard layout as an unpriviledged user

CutterXYZ, 2025-08-31:
Duplicate of issue [182](https://github.com/raspberrypi/bookworm-feedback/issues/182)

Unprivileged users cannot change their keyboard layout. The keyboard setting in Preferences shows the system-wide layout instead of the user layout which is confusing.

Changing this basic option requires Googling and manually editing hidden files.

Please reconsider this issue as a fixable bug.

lurch, 2025-08-31:
ping @spl237 for comment, but I believe the reason that changing he keyboard layout requires privileged access is because it also changes the keyboard layout used by the greeter program (i.e. the screen you see if you select "Logout", or if you disable "Auto login to desktop").

In Labwc (which is now used instead of Wayfire) you can find the config in `.config/labwc/environment` (which you can obviously edit as an unprivileged user). If you want to dive into what's _actually_ happening when you change the keyboard layout in preferences, see https://github.com/RPi-Distro/raspi-config/blob/bookworm/raspi-config#L785 and https://github.com/RPi-Distro/raspi-config/blob/bookworm/raspi-config#L676

spl237, 2025-09-01:
> ping [@spl237](https://github.com/spl237) for comment, but I believe the reason that changing he keyboard layout requires privileged access is because it also changes the keyboard layout used by the [...]

Correct.

CutterXYZ, 2025-09-01:
@lurch the use case for b) is to realign the user's keyboard layout with the system-wide layout. For an unknown reason, the users that were created with `useradd` had their keyboard layout set to a default layout instead of the system-wide layout.

spl237, 2025-09-01:
Running "sudo raspi-config nonint update_labwc_keyboard" will copy the current system keyboard config to the current user.

spl237, 2025-09-01:
It does require sudo, because as before, it also updates the greeter config file.

If someone wanted to just update the user config, they need to look at that function and pull out the relevant bits of it.

If you are creating new users using the system useradd or adduser functions, you will need to configure their keyboards under labwc as part of the process, in the same way as you will need to add them to appropriate groups and so on. But if you are creating new users, you are surely doing that as sudo anyway, so can set up their keyboard at the same time....

CutterXYZ, 2025-09-01:
@spl237 could this be automated with the adduser function ?

CutterXYZ, 2025-09-01:
Does this mean that Debian has the same problem with newly created users ?

CutterXYZ, 2025-09-01:
SHouldn't the labwc default config depend on the system setting ?

spl237, 2025-09-01:
That's a decision made by the authors of labwc, and is outside the scope of this issue.

https://labwc.github.io/labwc-config.5.html#environment_variables